csete/gpredict #368: New release

AsciiWolf, 2024-09-19:
Any chance a new Gpredict version could be tagged? It has been over 6 years since the last one was released.

/cc @csete

Hellseher, 2024-11-14:
Hi,

A gentle bump, it would allow to bring a fresh version to Guix astro package index.

Thanks

GideonWolfe, 2024-12-29:
Also bumping, just installed on NixOS and was confused why it was pulling a version from 2018 but it's the most recent in the releases... Would love to be able to change the background color of the polar module lol

Pandapip1, 2025-01-17:
Bump! I'm actually in the process of trying to modernize some of the gpredict stuff in nixpkgs (e.g. using goocanvas 3) and discovered that the last release, again, was from 2018. There have been 134 commits since then, and I'd rather not have to backport a bunch of stuff.

AsciiWolf, 2025-06-11:
@csete Any progress?