csete/gpredict #373: Communication issue between Gpredict and rotctld with Easycomm II (SARCNET) rotator.

nezumi-tech, 2025-02-06:
### What I tried
I am new to satellite communications and trying to utilize Gpredict and [SARCNET Mini Satellite-Antenna Rotator](https://www.sarcnet.org/rotator-mk1.html) Mk1 via [Hamlib](https://github.com/Hamlib/Hamlib).
I partially built a Rotator with Raspberry Pi Pico and sensor, without motor and mechanics, then followed [SARCNET's instructions ](https://www.sarcnet.org/rotator-mk1.html#InstallingGpredictHamlib) to setup Gpredict and Hamlib.

Configurations of my computer and software are below:
・OS : Windows 10 22H2
・Gpredict : 2.2.1
・Hamlib : w64-4.6.1

Note that [issue on serial communication](https://www.sarcnet.org/rotator-mk1.html#InstallingGpredictHamlib) is already reported by SARCNET, therefore  I used FTDI232 USB-Serial adapter to link the computer and the rotator.
Then, I found that there is issue when Gpredict communicates with rotator.

### Main issue

Probably there is a issue with the communication between Gpredict and rotctld.

### How to reproduce

Code on Raspberry Pi Pico is very simple: it listens on serial port and when command comesin, it does the following.

- "AZ" : Return "AZ123.45" (0 to 360) value.
- "EL" : Return "EL67.89" (0 to90) value.
- "AZ EL" : Return "AZ123.45 EL67.89" value.
- "AZ123.45 EL67.89" : no return.

I used following tracking.bat file that is based on ./share/tracking.txt .
```
set ROTCTLD=C:\"Program Files"\hamlib-w64-4.6.1\bin\rotctld.exe
set GPREDICT=./gpredict.exe
set ROTTYPE=202
set ROTPORT=COM10
set ROTRATE=9800
start /min "Rotator Control" %ROTCTLD% -m %ROTTYPE% -r %ROTPORT% -s %ROTRATE%  -T 127.0.0.1 -t 4533 -vvvvvv
start /wait %GPREDICT%
taskkill /f /im rotctld.exe
```

1. Connect the rotator to PC.
2. Run tracking.bat .
3. Open Gpredict Rotator Control.
4. Select satellite at range then push "Track" button.
5. Select configured rotator then push "Engage" button.
6. rotctld window shows following log, and Gpredict freezes.

From the log, it is seemed that Gpredict could get AZ/EL data only the first time, before stopped.

```
2025-02-07T00:36:36.950136+0900: rotctld, Hamlib 4.6.1 2025-01-21T09:43:13Z SHA=cb77f3 64-bit
2025-02-07T00:36:36.961637+0900: Report bugs to <hamlib-developer@lists.sourceforge.net>

2025-02-07T00:36:36.972137+0900: rot_init called
2025-02-07T00:36:36.978137+0900: initrots4_easycomm called
2025-02-07T00:36:36.985637+0900: rot_register (201)
2025-02-07T00:36:36.990638+0900: rot_register (202)
2025-02-07T00:36:36.996636+0900: rot_register (204)
2025-02-07T00:36:36.999637+0900: rot_open called
2025-02-07T00:36:37.121156+0900: serial_open: serial port COM10 is OK
2025-02-07T00:36:37.126642+0900: serial_open: COM10
2025-02-07T00:36:37.133136+0900: serial_setup: tcgetattr
2025-02-07T00:36:37.136637+0900: serial_setup: cfsetispeed=9600,0x000d
2025-02-07T00:36:37.140639+0900: serial_setup: cfsetospeed=9600,0x000d
2025-02-07T00:36:37.144637+0900: serial_setup: data_bits=8
2025-02-07T00:36:37.148637+0900: serial_setup: stopbits=1
2025-02-07T00:36:37.153138+0900: serial_setup: parity=0
2025-02-07T00:36:37.157136+0900: serial_setup: Handshake=None
2025-02-07T00:36:37.161137+0900: serial_setup: tcsetattr TCSANOW
2025-02-07T00:36:37.214639+0900: ser_set_dtr: DTR=0
2025-02-07T00:36:37.217637+0900: ser_set_rts: RTS=0
Opened rot model 202, 'EasycommII'
2025-02-07T00:36:37.223139+0900: Backend version: 20231218.0, Status: Stable
[... 142 lines cut ...]
2025-02-07T00:50:42.685138+0900: easycomm_transaction called: AZ

2025-02-07T00:50:42.689136+0900: write_block(): TX 3 bytes
2025-02-07T00:50:42.692138+0900: 0000    41 5a 0a                                            AZ.
2025-02-07T00:50:42.697638+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T00:50:42.716137+0900: read_string_generic(): RX 9 characters, direct=1
2025-02-07T00:50:42.721636+0900: 0000    41 5a 32 37 32 2e 39 0d 0a                          AZ272.9..
2025-02-07T00:50:42.726638+0900: easycomm_transaction read_string: AZ272.9

2025-02-07T00:50:42.730138+0900: easycomm_rot_get_position got response: AZ272.9

2025-02-07T00:50:42.734137+0900: easycomm_transaction called: EL

2025-02-07T00:50:42.737638+0900: write_block(): TX 3 bytes
2025-02-07T00:50:42.740639+0900: 0000    45 4c 0a                                            EL.
2025-02-07T00:50:42.745637+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T00:50:42.764137+0900: read_string_generic(): RX 8 characters, direct=1
2025-02-07T00:50:42.769136+0900: 0000    45 4c 2d 30 2e 32 0d 0a                             EL-0.2..
2025-02-07T00:50:42.775140+0900: easycomm_transaction read_string: EL-0.2

2025-02-07T00:50:42.778641+0900: easycomm_rot_get_position got response: EL-0.2

2025-02-07T00:50:42.782636+0900: rot_get_position: got az=272.90, el=-0.20
2025-02-07T00:50:50.524136+0900: Connection closed from 127.0.0.1:63446
```

mdblack98, 2025-02-06:
Get rid of the carriage return.  Only 1 char separating things.

 
 

   

   
   - Run tracking.bat .

p
320.10
146.70
P123.45 67.89
RPRT 0
p
342.90
157.10
p
272.90
-0.20
q

rotctld log
PS C:\Program Files\hamlib-w64-4.6.1\bin> ./rotctld.exe -m 202 -r COM10 -s 9600 -T 127.0.0.1 -t 4533 -vvvvvv -Z
2025-02-07T00:49:56.381140+0900: rotctld, Hamlib 4.6.1 2025-01-21T09:43:13Z SHA=cb77f3 64-bit
2025-02-07T00:49:56.386164+0900: Report bugs to ***@***.***>

2025-02-07T00:49:56.391140+0900: rot_init called
2025-02-07T00:49:56.393636+0900: initrots4_easycomm called
2025-02-07T00:49:56.396138+0900: rot_register (201)
2025-02-07T00:49:56.398637+0900: rot_register (202)
2025-02-07T00:49:56.401136+0900: rot_register (204)
2025-02-07T00:49:56.403137+0900: rot_open called
2025-02-07T00:49:56.532136+0900: serial_open: serial port COM10 is OK
2025-02-07T00:49:56.535136+0900: serial_open: COM10
2025-02-07T00:49:56.544136+0900: serial_setup: tcgetattr
2025-02-07T00:49:56.547137+0900: serial_setup: cfsetispeed=9600,0x000d
2025-02-07T00:49:56.550137+0900: serial_setup: cfsetospeed=9600,0x000d
2025-02-07T00:49:56.553638+0900: serial_setup: data_bits=8
2025-02-07T00:49:56.556640+0900: serial_setup: stopbits=1
2025-02-07T00:49:56.559640+0900: serial_setup: parity=0
2025-02-07T00:49:56.562639+0900: serial_setup: Handshake=None
2025-02-07T00:49:56.566137+0900: serial_setup: tcsetattr TCSANOW
2025-02-07T00:49:56.619637+0900: ser_set_dtr: DTR=0
2025-02-07T00:49:56.623139+0900: ser_set_rts: RTS=0
Opened rot model 202, 'EasycommII'
2025-02-07T00:49:56.627636+0900: Backend version: 20231218.0, Status: Stable
2025-02-07T00:49:59.351138+0900: Connection opened from 127.0.0.1:63446
2025-02-07T00:50:01.923136+0900: rotctl(d): p '' '' '' ''
2025-02-07T00:50:01.927136+0900: rot_get_position called
2025-02-07T00:50:01.931638+0900: easycomm_rot_get_position called
2025-02-07T00:50:01.935136+0900: easycomm_transaction called: AZ

2025-02-07T00:50:01.939136+0900: write_block(): TX 3 bytes
2025-02-07T00:50:01.942636+0900: 0000    41 5a 0a                                            AZ.
2025-02-07T00:50:01.949636+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T00:50:01.960136+0900: read_string_generic(): RX 9 characters, direct=1
2025-02-07T00:50:01.966137+0900: 0000    41 5a 33 32 30 2e 31 0d 0a                          AZ320.1..
2025-02-07T00:50:01.972636+0900: easycomm_transaction read_string: AZ320.1

2025-02-07T00:50:01.977137+0900: easycomm_rot_get_position got response: AZ320.1

2025-02-07T00:50:01.981638+0900: easycomm_transaction called: EL

2025-02-07T00:50:01.985636+0900: write_block(): TX 3 bytes
2025-02-07T00:50:01.989137+0900: 0000    45 4c 0a                                            EL.
2025-02-07T00:50:01.994636+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T00:50:02.007138+0900: read_string_generic(): RX 9 characters, direct=1
2025-02-07T00:50:02.011136+0900: 0000    45 4c 31 34 36 2e 37 0d 0a                          EL146.7..
[... 42 lines cut ...]
2025-02-07T00:50:42.681637+0900: easycomm_rot_get_position called
2025-02-07T00:50:42.685138+0900: easycomm_transaction called: AZ

2025-02-07T00:50:42.689136+0900: write_block(): TX 3 bytes
2025-02-07T00:50:42.692138+0900: 0000    41 5a 0a                                            AZ.
2025-02-07T00:50:42.697638+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T00:50:42.716137+0900: read_string_generic(): RX 9 characters, direct=1
2025-02-07T00:50:42.721636+0900: 0000    41 5a 32 37 32 2e 39 0d 0a                          AZ272.9..
2025-02-07T00:50:42.726638+0900: easycomm_transaction read_string: AZ272.9

2025-02-07T00:50:42.730138+0900: easycomm_rot_get_position got response: AZ272.9

2025-02-07T00:50:42.734137+0900: easycomm_transaction called: EL

2025-02-07T00:50:42.737638+0900: write_block(): TX 3 bytes
2025-02-07T00:50:42.740639+0900: 0000    45 4c 0a                                            EL.
2025-02-07T00:50:42.745637+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T00:50:42.764137+0900: read_string_generic(): RX 8 characters, direct=1
2025-02-07T00:50:42.769136+0900: 0000    45 4c 2d 30 2e 32 0d 0a                             EL-0.2..
2025-02-07T00:50:42.775140+0900: easycomm_transaction read_string: EL-0.2

2025-02-07T00:50:42.778641+0900: easycomm_rot_get_position got response: EL-0.2

2025-02-07T00:50:42.782636+0900: rot_get_position: got az=272.90, el=-0.20
2025-02-07T00:50:50.524136+0900: Connection closed from 127.0.0.1:63446

nezumi-tech, 2025-02-06:
> Get rid of the carriage return.  Only 1 char separating things.

Thank you. I fixed my Raspberry Pi Pico code like this. I missed Easycomm II protocol uses LF only.

```diff
if (line.startsWith("AZ") && line.length() == 2) {  //Query command received
    //Print the rotator feedback data in Easycomm II format
    SerialPort.print("AZ");
    SerialPort.print((az < 0) ? (az + 360) : az, 1);
-    SerialPort.println("");  // Send CRLF (0x0D 0x0A)
+    SerialPort.print("\n");  // Send LF (0x0A)
}
```

I found that when the rotator control “Monitor” was checked and “Engage” was pressed, Gpredict can keep reading the angle of the rotator forever.
However, when the “Track” button and then the “Engage” button are pressed, the situation is the same: the Gprerdict freezes after a few AZ/EL readings. According to the rotctld logs, Gpredict does not seem to be sending any messages to move the rotator such as "P123.45 67.89\n".

rotctld log
```
2025-02-07T04:12:58.533202+0900: rotctld, Hamlib 4.6.1 2025-01-21T09:43:13Z SHA=cb77f3 64-bit
2025-02-07T04:12:58.542203+0900: Report bugs to <hamlib-developer@lists.sourceforge.net>

2025-02-07T04:12:58.552702+0900: rot_init called
2025-02-07T04:12:58.557702+0900: initrots4_easycomm called
2025-02-07T04:12:58.564202+0900: rot_register (201)
2025-02-07T04:12:58.569201+0900: rot_register (202)
2025-02-07T04:12:58.574205+0900: rot_register (204)
2025-02-07T04:12:58.577702+0900: rot_open called
2025-02-07T04:12:58.707705+0900: serial_open: serial port COM10 is OK
2025-02-07T04:12:58.714705+0900: serial_open: COM10
2025-02-07T04:12:58.735202+0900: serial_setup: tcgetattr
2025-02-07T04:12:58.739201+0900: serial_setup: cfsetispeed=9600,0x000d
2025-02-07T04:12:58.743737+0900: serial_setup: cfsetospeed=9600,0x000d
2025-02-07T04:12:58.747703+0900: serial_setup: data_bits=8
2025-02-07T04:12:58.751202+0900: serial_setup: stopbits=1
2025-02-07T04:12:58.755202+0900: serial_setup: parity=0
2025-02-07T04:12:58.758204+0900: serial_setup: Handshake=None
2025-02-07T04:12:58.761728+0900: serial_setup: tcsetattr TCSANOW
2025-02-07T04:12:58.815701+0900: ser_set_dtr: DTR=0
2025-02-07T04:12:58.818703+0900: ser_set_rts: RTS=0
Opened rot model 202, 'EasycommII'
2025-02-07T04:12:58.824202+0900: Backend version: 20231218.0, Status: Stable
2025-02-07T04:13:19.176203+0900: Connection opened from 127.0.0.1:64868
2025-02-07T04:13:19.276703+0900: rotctl(d): p '' '' '' ''
2025-02-07T04:13:19.280202+0900: rot_get_position called
2025-02-07T04:13:19.283701+0900: easycomm_rot_get_position called
2025-02-07T04:13:19.287702+0900: easycomm_transaction called: AZ

2025-02-07T04:13:19.292202+0900: write_block(): TX 3 bytes
2025-02-07T04:13:19.296202+0900: 0000    41 5a 0a                                            AZ.
2025-02-07T04:13:19.303203+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T04:13:19.314201+0900: read_string_generic(): RX 8 characters, direct=1
2025-02-07T04:13:19.318702+0900: 0000    41 5a 31 34 38 2e 39 0a                             AZ148.9.
2025-02-07T04:13:19.326204+0900: easycomm_transaction read_string: AZ148.9

2025-02-07T04:13:19.331203+0900: easycomm_rot_get_position got response: AZ148.9

2025-02-07T04:13:19.336203+0900: easycomm_transaction called: EL

2025-02-07T04:13:19.341204+0900: write_block(): TX 3 bytes
2025-02-07T04:13:19.344703+0900: 0000    45 4c 0a                                            EL.
2025-02-07T04:13:19.351701+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T04:13:19.361703+0900: read_string_generic(): RX 8 characters, direct=1
2025-02-07T04:13:19.367201+0900: 0000    45 4c 31 36 36 2e 34 0a                             EL166.4.
2025-02-07T04:13:19.373704+0900: easycomm_transaction read_string: EL166.4

2025-02-07T04:13:19.378702+0900: easycomm_rot_get_position got response: EL166.4

2025-02-07T04:13:19.384202+0900: rot_get_position: got az=148.90, el=166.40
2025-02-07T04:13:20.189703+0900: rotctl(d): p '' '' '' ''
[... 28 lines cut ...]
2025-02-07T04:13:21.142701+0900: easycomm_transaction called: AZ

2025-02-07T04:13:21.147702+0900: write_block(): TX 3 bytes
2025-02-07T04:13:21.151702+0900: 0000    41 5a 0a                                            AZ.
2025-02-07T04:13:21.159202+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T04:13:21.168203+0900: read_string_generic(): RX 8 characters, direct=1
2025-02-07T04:13:21.173202+0900: 0000    41 5a 31 34 38 2e 34 0a                             AZ148.4.
2025-02-07T04:13:21.180202+0900: easycomm_transaction read_string: AZ148.4

2025-02-07T04:13:21.184703+0900: easycomm_rot_get_position got response: AZ148.4

2025-02-07T04:13:21.190705+0900: easycomm_transaction called: EL

2025-02-07T04:13:21.195202+0900: write_block(): TX 3 bytes
2025-02-07T04:13:21.199202+0900: 0000    45 4c 0a                                            EL.
2025-02-07T04:13:21.206703+0900: read_string_generic called, rxmax=32 direct=1, expected_len=1
2025-02-07T04:13:21.213201+0900: read_string_generic(): RX 8 characters, direct=1
2025-02-07T04:13:21.218202+0900: 0000    45 4c 31 36 36 2e 31 0a                             EL166.1.
2025-02-07T04:13:21.227202+0900: easycomm_transaction read_string: EL166.1

2025-02-07T04:13:21.232702+0900: easycomm_rot_get_position got response: EL166.1

2025-02-07T04:13:21.237702+0900: rot_get_position: got az=148.40, el=166.10
```
(Then stopped.)

nezumi-tech, 2025-02-06:
> Don't use the Monitor checkbox -- that stops it sending.

Yes, I know. So I opened the Rotator Control window, then pressed Track and Engage button. But Gpredict doesn't send AZ/EL target data.

[image]

mdblack98, 2025-02-06:
Works fine here so don't know what your problem is....

 
 

image.png (view on web)

nezumi-tech, 2025-02-06:
The problem is the rotator doesn't move. As we can see, in the rotcrld window, debug messages are enabled and should show "easycomm_rot_set_position called: XXX.X XX.X", but it does not. So, it seems that Gpredict does not call easycomm_rot_set_position() in [Hamlib/rotators/easycomm/easycomm.c ](https://github.com/Hamlib/Hamlib/blob/c939f1a83ac1b7ad1bb903120da13f19b468e4be/rotators/easycomm/easycomm.c#L103).  Then the command to set AZ/EL such as "AZ120.62 EL25.03" is not sent to the rotator, therefore rotator can't move.

[image]

nezumi-tech, 2025-02-08:
I confirmed that no error occurs when running both Gpredict and rotctld on a Raspberry Pi 3+.
The left window is Raspberry Pi Connect remote desktop and the right window is SSH console to the same Raspberry Pi.
As we can see, AZ/EL control command such as "P 254.43 3.59" is sent from Gpredict to rotctld, then command "AZ254.4 EL3.6" are sent to the rotator. Furthermore, AZ/EL querying "p" and "AZ EL" commands are also transmitted then the value is displayed on Gpredict.
So I think the problem with this issue is specific to Windows.

[image]

mdblack98, 2025-02-08:
Hamlib version on each?GPredict version on each?

nezumi-tech, 2025-02-08:
> Hamlib version on each?GPredict version on each?

My desktop PC
・OS : Windows 10 22H2
・Gpredict : 2.2.1
・Hamlib : w64-4.6.1

Raspberry Pi 3B+
・Gpredict : 2.3-72-gc596101-3 
Installed by apt install gpredict

[image]

・Hamlib :
 $ rotctl -V
rotctl(d), Hamlib 4.5.4 Jan 10 01:31:41Z 2023 SHA=921cc5
I downloaded source hamlib-4.6.1.tar.gz from https://github.com/Hamlib/Hamlib/releases/tag/4.6.1 and I ran make and make install.

nezumi-tech, 2025-02-08:
Oh! It is success! I could get good serial communication. The problem might be that I used old Gpredict versions. But can I ask you how we can get new versions of Gpredict?
[image]

For reference, the successful environment is below:
My desktop PC
・OS : Windows 10 22H2
・Gpredict : 2.3.115 I downloaded above.
・Hamlib : w64-4.6.1 from https://github.com/Hamlib/Hamlib/releases/tag/4.6.1

This is my hardware. The actuator is my hand.
[image]
PC <-> FTDI232 <-> RasPi Pico <-> LSM303DLH.

Note that if the PC and RasPi Pico are connected directly with a USB cable, serial communication on rotctld will not work properly.

mdblack98, 2025-02-08:
Updates to gpredict don't happen very often.
The version you have now is one that I built.
Mike W9MDB

 
 

image.png (view on web)

image.png (view on web)

nezumi-tech, 2025-02-09:
I see. Then the problem is solved, so I close this issue. Thank you for your kind support.

closed 2025-02-09 by nezumi-tech